igrr/axtls-8266 #2: Add an option for static or pool-based memory allocation

igrr, 2015-11-26:
Necessary because heap fragmentation becomes an issue as seen here: https://github.com/esp8266/Arduino/issues/1084

liquidfalcon, 2016-02-25:
@igrr Is it possible to handle an allocation failure gracefully in axtls, at least as a workaround for now until static allocation is worked through? I'm hitting the same issue as everyone else with SSL and realloc, but, I was curious if we could handle a nullptr being given by realloc, instead of just calling panic.

igrr, 2016-02-26:
@liquidfalcon thanks for the suggestion, i have implemented the workaround you described.

slaff, 2016-04-13:
> heap fragmentation becomes an issue

May be you can integrate [this project](https://github.com/rhempel/umm_malloc) for better heap allocation? Here is an article about it https://blog.cesanta.com/embedded-heap-behaviour-analysis-and-improvement.

igrr, 2016-04-13:
ESP866 Arduino core uses umm_malloc since last release.

slaff, 2016-04-13:
> ESP8266 Arduino core uses umm_malloc since last release.

Cool! And is it working as expected: Less heap fragmentation, etc?

igrr, 2016-04-13:
Yes, it's much better now. Although some fragmentation is still inevitable, and requesting 16k TLS buffer fails occasionally. Umm_malloc is slightly slower, and even more so if heap poisoning is enabled.