igrr/axtls-8266 #10: Not a valid ASN.1 file

slaff, 2016-03-03:
When integrating the code in  [another ESP8266 framework](https://github.com/SmingHub/Sming/pull/596) I am getting the following error before handshake.

```
Error: This is not a valid ASN.1 file
Error: Invalid X509 ASN.1 file (X509 not ok)
```

What can be the reason for this? With the Linux axtls code (without additions from this repo) it seems to work. Any suggestion?

igrr, 2016-03-03:
Which server should I use to reproduce this?

slaff, 2016-03-03:
I used the following [key](https://github.com/slaff/Sming/blob/feature/axtls-dev/samples/Basic_Ssl/include/ssl/private_key.h) and [certificate] ( https://github.com/slaff/Sming/blob/feature/axtls-dev/samples/Basic_Ssl/include/ssl/cert.h ) and I got this response testing with www.google.com and a lot of other servers. Actually I do not remember not getting this error.

igrr, 2016-03-03:
I wonder why would you need client certificate when you connect to google.com... If the server doesn't request client certificate it is not used.
Anyway, I'll check and see what's the issue with certificate loading.

noelgeorgi, 2016-03-18:
is this because the client certificate was generated without **-addtrust clientAuth** in openssl?

slaff, 2016-04-08:
@igrr @noelgeorgi  It turned out that the key and certificate that I was passing to `ssl_obj_memory_load` were in PEM format, whereas axTLS expects them at that stage to be in DER format. Passing the data in DER format fixed the issue.

closed 2016-04-08 by slaff