igrr/axtls-8266 #11: Merge new functions into the upstream axtls repository

slaff, 2016-03-11:
There are new and useful functions  that are appearing in this repository. It would be great if we can push them to the upstream axTLS repository.

igrr, 2016-03-15:
That would probably be a good effort, although I imagine it may take quite a lot of resources.
We should rebase our changes on the latest upstream version first (#9).

mirko, 2019-03-24:
Is there any news on this?
I'm asking specifically, because I'm running out of memory with mbedtls on ESP32 (using micropython*) and, from evaluating code and guessing, I expect axtls being much more lightweight.
However, unlike this fork, upstream axtls does not support verification against a CA so far, which is a major drawback.
Is there any plan on upstreaming those changes? If not, could you provide a rough estimate on how much is ESP8266 specific and might/will conflict?

Thanks a lot in advance!

*micropython supports both mbedtls as well as axtls as ssl backend

igrr, 2019-03-25:
@mirko On the ESP32, you can enable CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN (https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-mbedtls-asymmetric-content-len) to reduce the amount of RAM used by the outgoing fragment buffer.