igrr/axtls-8266 #12: ssl->need_bytes issue

noelgeorgi, 2016-03-17:
@igrr i am still getting **ssl->neeed bytes** error. I changed the value of 
**ssl->max_plain_length** to **ssl->max_plain_length = 6000*4;** which I think is the maximum value, i can give. Here is the dump: 

**~ld
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
chg_B1:-40
..
connected with TechnoMan, channel 6
ip:172.16.0.120,mask:255.255.255.0,gw:172.16.0.1
.IP: 172.16.0.120
Successfully opened ca file
loaded
IP: 172.16.0.120
Thu Mar 17 19:56:05 2016

Error: Invalid X509 ASN.1 file (X509 not ok)
pm open,type:2 0
IP: 172.16.0.120
Error: Invalid X509 ASN.1 file (X509 not ok)
ssl->need_bytes=37663 > 25019
IP: 172.16.0.120
Error: Invalid X509 ASN.1 file (X509 not ok)
IP: 172.16.0.120
Error: Invalid X509 ASN.1 file (X509 not ok)
ssl->need_bytes=61896 > 25019
IP: 172.16.0.120
Error: Invalid X509 ASN.1 file (X509 not ok)
ssl->need_bytes=52978 > 25019
IP: 172.16.0.120
Error: Invalid X509 ASN.1 file (X509 not ok)
IP: 172.16.0.120
Error: Invalid X509 ASN.1 file (X509 not ok)
ssl->need_bytes=26851 > 25019
**

igrr, 2016-03-17:
Looks like the other side is not talking over SSL. Per SSL spec, fragment size can not exceed 16 kilobytes, but your log shows that message header indicates size >16k (37663, 61896, 52978, 26851).

noelgeorgi, 2016-03-18:
@igrr what about **Error: Invalid X509 ASN.1 file (X509 not ok)**

igrr, 2016-03-18:
That one is open in #10

noelgeorgi, 2016-03-18:
@igrr this is the error i receive at the server side:
**1458292496: OpenSSL Error: error:14094410:SSL routines:SSL3_READ_BYTES:sslv3 alert handshake failure**

igrr, 2016-03-18:
Could you please give me an URL of the server so I can try myself?

noelgeorgi, 2016-03-18:
it is an MQTT broker (mosquitto) running at frezbo.com on port 8883, I can succesfully connect to it using MQTTfx a client for linux, how do i PM you the login details and cert files

igrr, 2016-03-18:
Okay, i don't think i need login details or cert files to establish a TLS connection. Obviously the server will kick me out later on, but your issue happens during handshake, to that would be enough.

noelgeorgi, 2016-03-19:
@igrr I changed the MQTT library i was using from PubSubClient to Adafruit's MQTT library and now there seems to be no issue till now. The funny fact is that loading or not loading the CA cert have no effect whatsoever.

aknik, 2018-05-06:
Same error here :-((

webSocket.beginSSL("dcf77logs.de", 443,"/ajax/liveview");

[WSc] Disconnected!
please start sntp first !
Error: Invalid X509 ASN.1 file (X509 not ok)

aknik, 2018-05-06:
With >>>  webSocket.beginSSL("dcf77logs.de", 80,"/ajax/liveview");

please start sntp first !
ssl->need_bytes=20527 violates spec
[WSc] Disconnected!