igrr/axtls-8266 #28: Licensing clarification

makedin, 2016-11-05:
I have not quite been able to figure out whether this repository falls under the 3-clause BSD lisence, or if that is just the upstream's lisence. The GitHub UI implies that the BSD lisence is indeed in use, but it might just automatically derive that from the LISENCE file.

pfalcon, 2016-11-13:
@igrr : Put it another way, people of the world would appreciate explicit confirmation that everything in this repository still falls under the original axTLS license, which is BSD. While everyone knows that this project is developed for the purposed of being included in https://github.com/esp8266/Arduino , which is LGPL-licensed. Thanks.

igrr, 2016-11-14:
Sorry for the confusion. I thought that 'LICENSE' file in the repository root is as clear as it gets.
In terms of licensing, this project is not related to Arduino, and it has the same license as the upstream project. In fact, given that Cameron Rich is the copyright owner, I don't see myself distributing this under any other license, even if I wanted to. Thanks.

closed 2016-11-14 by igrr

pfalcon, 2016-11-14:
@igrr: Thanks for the response. The LICENSE file should be enough for any open-source professional, like you and me. Unfortunately, many people and organizations don't understand open-source licensing well enough (for example, your employer still doesn't completely grok it), so it's a good idea to ask in any case which even remotely can lead to confusion. Thanks again for affirmative response.

> In fact, given that Cameron Rich is the copyright owner, I don't see myself distributing this under any other license, even if I wanted to.

Well, he's copyright owner for his code, and you're copyright owner for your code, and his license doesn't disallow you to re-license complete library with your modifications under another license. Of course, nobody would do that out of top of their head, but e.g. being paid to do just that would be a good reason. 

So, my plan for one of these months would be to look at the SNI implementation you wrote, and integrate into MicroPython's fork, and maybe even post upstream (I agree that takes resources, and I'm short of them now too, and axTLS isn't the most responsive upstream).

igrr, 2016-11-14:
> and his license doesn't disallow you to re-license complete library with your modifications under another license

Okay, point taken.

> So, my plan for one of these months would be to look at the SNI implementation you wrote, and integrate into MicroPython's fork, and maybe even post upstream.

Great, let me know if I you need my help on this. Sadly we have accumulated a fair amount of duct tape fixes and workarounds for memory issues in this fork... But SNI part shouldn't have much overlap with these random changes, except for maybe the place where the context is cleaned up.

> for example, your employer still doesn't completely grok it
Probably OT, but if there are some issues which you care about, please do send me an email. Thanks.