igrr/axtls-8266 #39: This lib is behind original codebase by several versions

ADiea, 2017-02-25:
Hi 
This lib is based on 1.5.3 as I see but the current code is at 2.1.3
I can start making a PR to get all modifications but I see little activity on this repo. 

@igrr if I start a PR to rebase this to latest axtls do you think you will have time to review and merge?
Thank you!

igrr, 2017-02-25:
We actually had 2.0 merged in, but the readme wasn't updated. If you rebate upon upstream I will surely review and merge this.
I have started testing your other PRs with Arduino and had some issues but haven't had time to figure out the problem yet. Probably next week...

ADiea, 2017-02-25:
Thanks for the answer will start a rebase soon

closed 2017-02-25 by ADiea

ADiea, 2017-03-01:
@igrr also if you provide more details i can invest some time to see why you experience issues with my other PRs :)