igrr/axtls-8266 #42: Compiling and using current version crashes ESP after HTTPS connection

dandandandandandandan, 2017-03-23:
Running the HTTPS example sketch results in a crash after connection when i clone and compile and then use the up to date libaxtls.a file.

Serial output:

WiFi connected
IP address: 
192.168.0.101
connecting to iot.thinger.io
certificate matches
requesting URL: /repos/esp8266/Arduino/commits/master/status
request sent
esp8266/Arduino CI has failed
reply was:
==========

==========
closing connection

Exception (3):
epc1=0x4010010e epc2=0x00000000 epc3=0x00000000 excvaddr=0x40046f69 depc=0x00000000

ctx: cont 
sp: 3fff1340 end: 3fff15d0 offset: 01a0

>>>stack>>>
3fff14e0:  3fff0558 000007a1 000007a1 4010020c  
3fff14f0:  3fff0558 0000aa2f 0000aa2f 4010020c  
3fff1500:  3ffe9314 00000000 3ffe9514 4010068c  
3fff1510:  3ffe9314 00000000 3fff2a84 4022e58d  
3fff1520:  3ffe9314 00000000 3fff2a84 4022faa7  
3fff1530:  3ffe9314 00000000 3fff2984 40203026  
3fff1540:  3ffe9458 3ffe94b8 3fff1560 40203074  
3fff1550:  3ffe9314 3ffe94b8 3fff0578 402021e6  
3fff1560:  3ffe97f8 00000000 000003e8 0000164e  
3fff1570:  00000000 3fff283c 3fff2984 00000000  
3fff1580:  00000000 00000000 00000000 00000000  
3fff1590:  00000000 00000000 00000000 00000000  
3fff15a0:  feefeffe feefeffe feefeffe 3fff05a4  
3fff15b0:  3fffdad0 00000000 3fff059c 40204138  
3fff15c0:  feefeffe feefeffe 3fff05b0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

igrr, 2017-04-29:
The current version of axTLS library is not compatible with the Arduino core, because the prototype of `ssl_client_new` function has changed.

igrr, 2017-04-29:
Should be fixed now.

closed 2017-04-29 by igrr