igrr/axtls-8266 #56: tls_verify always returns -4 [X509_VFY_ERROR_NOT_YET_VALID]

polandj, 2018-02-18:
Arduino IDE: 1.8.5
ESP platform version: 2.4.0
Device: ESP-12 (nodemcuv2)

I'm trying to get my ESP to connect to AWS IoT.  Using the attached sketch based on [this one](https://github.com/copercini/esp8266-aws_iot/blob/master/examples/MQTT_x509_bin/MQTT_x509_bin.ino), I always get that the certificate is not yet valid:

> Attempting MQTT connection...failed, rc=-4 try again in 5 seconds
> Heap: 20376
> TIME: 1518979598

I have [attached](https://github.com/igrr/axtls-8266/files/1734986/aws-iot.tar.gz) (gzipped, sorry, GH doesn't like all extensions) the sample sketch and certificates in PEM and DER format.  These are valid certs and endpoints, I know it's a security issue and plan on revoking/replacing them once I've figured this out.  

I'm 90% certain this is a user error, but I can't seem to figure it out.  I've tried editing the axtls library to print out the supposedly old timestamp versus the certificate span, but I can't figure out how to get the library to actually be linked against.   It seems to be, but clearly doesn't take my changes.  What I did:

1.  Cloned https://github.com/igrr/axtls-8266.git
2. Edit ssl/x509.c
3. Got it to compile, with new library in bin/libaxtls.a
4. Copied the new library to ~/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lib.
5. Recompiled (tried both from IDE and via makeEspArduino.mk) and flashed.

So, not sure what else to do to debug.

polandj, 2018-02-23:
For anyone else that comes across this.  It seems that instead, if you create your own CA and certificates, you can then get it to connect.  However, be aware that on the ESP8266, the initial handshake negotiation can take between 5 and 15 seconds.  You'll need to deal with the watchdogs, etc.  Learn more here:
https://github.com/esp8266/Arduino/issues/3944

closed 2018-02-23 by polandj

slaff, 2018-02-26:
>  the initial handshake negotiation can take between 5 and 15 seconds.

@polandj You might be doing something wrong. From my humble experience the handshake takes less than 300 ms. Make sure to optimize your certificates on the server side and remove things like Netscape comments or other useless for a production certificate data.  Although I am using the same library integrated in a different framework called Sming (https://github.com/SmingHub/Sming/tree/develop/samples/Basic_Ssl) I highly doubt that under Arduino that same library will work much slower.